frederic-mahe/Analogue-Pocket #1: EZ Flash Omega Definitive Edition compatibility issues

coolbho3k, 2021-12-26:
Three possible issues with the EZ Flash Definitive Edition have been reported, in decreasing order of severity:

- [Cart won't boot at all. Analogue BIOS Link Error.](https://gbatemp.net/threads/ez-flash-omega-analogue-pocket-firmware.604607/)
- [Cart boots games, but crashes occasionally](https://www.reddit.com/r/AnalogueInc/comments/rorm08/compatibility_on_the_ez_flash_omega_de_and_the/) .
- Cart boots in Mode A and games work well, but you get a white screen in Mode B (boot directly from NOR flash). Also cannot boot to a game in the NOR flash from the menu. For me, the EZ Flash in-game menu functionality for in-game reset and save states also freezes from time to time and doesn't freeze on a real GBA.

coolbho3k, 2021-12-29:
It's worth noting that the EZ Flash team has [requested a Pocket unit to test with](https://gbatemp.net/threads/ez-flash-omega-analogue-pocket-firmware.604607/#post-9695796).

It appears they're in fulfillment Group C. I think it's worth it to send them a development unit as they are one (of the two) most popular current flashcart manufacturers.

coolbho3k, 2022-01-01:
One person on GBAtemp got the Omega DE to work [by inserting a real cart, sleeping the Pocket, swapping to his flashcart, and unsleeping the Pocket](https://gbatemp.net/threads/ez-flash-omega-analogue-pocket-firmware.604607/post-9705901)

Can anyone else who is experiencing the "Analogue BIOS Link Mode" screen with the Omega DE confirm if this fixes it?

Malec1337, 2022-01-04:
I followed each step from the post you linked to and it worked for me. I was able to use my EZ Flash Omega DE this way.

coolbho3k, 2022-01-05:
> I followed each step from the post you linked to and it worked for me. I was able to use my EZ Flash Omega DE this way.

That's awesome. This means it's fixable in software.

By the way, can you save state on the Omega DE's menu and then successfully load it when you get the white screen again?

Malec1337, 2022-01-06:
Save State did not work for me. I started Bomberman 2 Blue on my DE with the steps mentioned earlier and went into the DE Save State menu (Start + L + R). The menu opened and i choose "save". After this the game was broken. The hole screen turned green. I was able to go into the Save State menu again and choose "load". After that my Pocket crashed. I had to reboot it.

coolbho3k, 2022-01-07:
Omega DE save states and in general in-game menu is a hit or miss for me and definitely is more finicky than real hardware.

I've gotten it to work well in Pokemon Emerald but it didn't work so well in Mother 3.

coolbho3k, 2022-01-16:
My Omega DE very suddenly stopped working with the Pocket today. It was working yesterday. I had left the cart in the Pocket and turned it on this afternoon and it wasn't working. I cleaned the contacts very thoroughly with contact cleaner and it still doesn't work.

Everything else seems to be fine.

The "sleep with a real game" workaround works to boot the Omega DE, but then it prompts me to update the kernel.

Malec1337, 2022-01-17:
Strange. So it is possible that it will not work in the future even if it is working now. I hope EZ Flash and Analogue are able to fix this soon. My EZ Flash Junior works great with the Pocket (Newest Beta Firmware) and i hope my Omega DE will soon. I like my EZ Flashes :)

coolbho3k, 2022-02-01:
1.0B has fixed the issue for me again. The Pocket lingers on the Analogue BIOS screen for a second unlike with other carts, leading me to believe that the firmware is now detecting the Omega DE and giving it a bit more time to initialize before booting. Anyone else want to try?

Boot from NOR doesn't work. Haven't tried Mode B yet.

Malec1337, 2022-02-02:
I tried it and it works for me now. As you mentioned it stays a little bit longer in the BIOS screen and then boots the Omega DE.

frederic-mahe, 2022-02-02:
> Boot from NOR doesn't work. Haven't tried Mode B yet.

hi @coolbho3k if you think the issue is solved, please close it. If the problem is partially solved, maybe it is best to open a new issue with a more specific description.

noj86, 2022-02-08:
I confirm that 1.0B brought issue with my Omega DE, I reverted back to 1.0A to make it work again but with the latest firmware I get the Bad Header error systematically. I have found at last one other person getting the exact same issue as me since the 1.0B update. https://twitter.com/JoeSpaghet/status/1488622060200992779

I've reported more specifically the issue in this thread and tested against a regular non DE Omega and the Ez-Flah Junior too.
https://gbatemp.net/threads/analogue-pocket-fw-1-0b-ez-flash-omega-de-new-issue-after-first-boot.607082/

coolbho3k, 2022-02-08:
I'm going to close this issue and make a new issue to separate out the 1.0B issues: #63

closed 2022-02-08 by coolbho3k

KeenIIDX, 2022-07-30:
Today’s 1.1 beta finally fixed my bad header errors. I have a working flash cart!