frederic-mahe/Analogue-Pocket #2: Everdrive GBA X5 compatibility issues

coolbho3k, 2021-12-26:
Some Everdrive GBA X5 units will fail to save with EEPROM games.

When EEPROM games fail, the Everdrive's built-in diagnostic test will usually fail on the EEPROM test as well. 

However, this works fine on real hardware, and only affects some GBA X5 units.

QuadPiece, 2021-12-27:
My GBA X5 (Non-mini) works mostly fine, but will freeze once in a while. Sometimes 5 minutes into a game, sometimes 30 minutes into a game. The game will simply freeze on a frame and the audio bug out.

Real and repro carts work fine so I don't think it's a connection issue with the slot. Issue does not happen on an AGS-101 or AGS-001.

schmingy, 2021-12-28:
Minish cap and A Link to the Past / Four Swords don’t save on the x5 mini I have but do on original cartridges.

coolbho3k, 2021-12-28:
> Minish cap and A Link to the Past / Four Swords don’t save on the x5 mini I have but do on original cartridges. 

Can you try the diagnostic test?

schmingy, 2021-12-28:
> > Minish cap and A Link to the Past / Four Swords don’t save on the x5 mini I have but do on original cartridges.
> 
> Can you try the diagnostic test?

The EEPROM came back with Error 1 when I did it this weekend on the pocket. 

The same Everdrive passes all the tests on the SP I have and the game saves and survives a power off, removal, reinsertion, and reload of the rom file.

16bitstreamer, 2021-12-29:
I can get Final Fantasy IV, & A Link to the Past to both save and load without issue using the x5 mini. Diagnostic is OK across all tests. OS Version 1.15.

coolbho3k, 2022-01-11:
OS Version 1.16 fixes this issue on the Everdrive side.

However, I'll leave this issue open as there's still a slight timing bug on Analogue's GBA core that may also be affecting EEPROM saves on the EZ Flash Omega.