frederic-mahe/Analogue-Pocket #3: Everdrive GB X7 and Everdrive GB X5 compatibility issues

coolbho3k, 2021-12-26:
Some Everdrive GB X7 and Everdrive GB X5 units will show an SD I/O error when trying to write to the SD card. This makes the flash cartridge almost unusable as it must write to the SD card when loading games and saving save games to the SD card.

This seems to be  almost completely unpredictable based on the Everdrive's manufacturing date and PCB revision. The current theory is it's dependent on subtle variations among each individual Everdrive unit. It may affect later units more than earlier units.

bbbbbr, 2021-12-26:
Can we amend this issue title to also include GB X3?

For me it's not common to get the SD Card error on the X3 (~2018-2019) vintage, but it does happen on occasion and mostly when browsing the SD card contents from the EverDrive menu. Once that happens it requires a quit + restart cartridge from the Pocket's menu.

GB X3 Model 17, Rev C, 20.02.2018

QuadPiece, 2021-12-27:
I have an Everdrive GB X3. Never seen the SD I/O error, but typically encounter graphical artifacts only a few minutes into playing any game, and then the game freezes. I'm not sure if it's the same issue, but it does mean I can't really play any games reliably via my GB X3

Doesn't happen on real hardware, and my Pocket has had no issue with real or repro carts.

KeenIIDX, 2021-12-27:
I think each model should be a separate Issue. Maybe they all have the same underlying issue, maybe they don’t. If they get fixed, folks can track their specific models/Issues better is they’re not jumbled into one pile.

BestPig, 2021-12-27:
> I think each model should be a separate Issue. Maybe they all have the same underlying issue, maybe they don’t. If they get fixed, folks can track their specific models/Issues better is they’re not jumbled into one pile.

I disagree, it's the same issue.
Read / Write on the SD card is unreliable, It's just that the x7 is more affected but graphics glitch on x3 is due to unreliable read.
People with an x7 that seems to work have often also graphics glitches like the x3.

So for me it's the same issue.

coolbho3k, 2021-12-28:
I agree that it still makes sense to keep this in one issue for now. Let's split it out once more is known or Analogue manages to fix one of the issues but not the others.

As far as I know the Everdrive X series all share the same fundamental design, with different enabled features and components soldered to the PCB.

QuadPiece, 2021-12-28:
> As far as I know the Everdrive X series all share the same fundamental design, with different enabled features and components soldered to the PCB.

I'm fairly certain this is the case. It might also be related to how common issues are. If the issue is related to power, it would make sense that the X7 is more likely experience issues due to having more components and thus drawing slightly more power, while the X3, with the lowest number of components, is least likely to experience issues.

I agree that all Everdrive GB X carts should be tracked in the same issue until we figure out if the cause is model specific or it's the same issue affecting the whole range.

coolbho3k, 2021-12-30:
People on the Discord have mentioned that pulling out the cart slightly (not inserting it all the way) may help with SD read/write errors, perhaps _suggesting_ some tolerance variations with cart slots.

BestPig, 2021-12-30:
I tried and nothing for me, still have the I/O issue.
And I tried a lot of different position.

tjanas, 2021-12-30:
Looks like Analogue released a firmware update this morning (v1.0A), does it resolve the EverDrive GB X-series issues?

zfleeman, 2021-12-30:
> Looks like Analogue released a firmware update this morning (v1.0A), does it resolve the EverDrive GB X-series issues?

It does not.

mchiaramonte, 2021-12-30:
I just wanted to add that I updated to firmware 1.0a just now and while I had weird and sporadic issues with the GB x3 OS and some games with firmware 1.0, it’s totally broken for me with 1.0a.

bender-the-greatest, 2021-12-30:
At this point I'm not sure where the issue lies. The problem EDx7s are working on OG hardware, but only sometimes with the Pocket. The timing fixes seem to have fixed the issue for some, not others, and some previously working EDx7s are being reported as no longer working with the 1.0A firmware. Yet even the latter issue is inconsistent.

If anyone with an Everdrive has access to more than one Pocket (I know some of us bought two for yourself or one for a friend), it would be good to know whether the issues occur across Pockets or if issues seem isolated to specific Everdrive units.

sssembler, 2021-12-30:
Before the update I had a x7 that worked perfectly and a x5 gba that worked perfectly. After still working great.

tkbryant, 2021-12-30:
Both of my Everdrive GBA X5 & GB X7 are working just fine both before & after the update. Maybe a stupid question but have others been keeping up with their firmware updates on their Everdrives?

mchiaramonte, 2021-12-31:
@tkbryant I had updated the firmware on my X3 on the off chance that it might help, but no luck. I did try a different SD card, too, but no change there. Both everdrives work fine in original hardware of all different “flavors”

16bitstreamer, 2021-12-31:
Okay, here’s a thought: I have a knock-off Everdrive that I got off of Amazon. Gives me a read I/O error when I try to use it. So, I noticed it was a little loose in how it sat in the pocket’s cartridge slot. So I slid the cart all the way to one side and tested again. Works now without issue except for the occasional reseating. (I don’t recommend this, as I’m afraid I’ll damage the Pocket, but, it works) 

I understand that it’s not an authentic krikzz cart, but, has anyone tried this? I updated the bootleg cart with official firmware though, latest revision of the GB Original GBOS.gb. 

My authentic Everdrive GB x7 & GBA x5 mini both work without any issues. 

Just throwing this out there, seeing if it helps.

bender-the-greatest, 2021-12-31:
This leads me to believe it could be an issue with the cartridge slot pinout on some units. I wonder...

Is there a correlation between the folks with non-working Everdrives who also have the following reported issues with **official** cartridges?

- Cartridge wiggles/or feels loose when in the slot.
- Bumping the cartridge gently borks the game (e.g. freeze, artifacts, etc.)
- Setting the Pocket down on a hard surface sometimes (or often) causes the prior issue
- Visibly misaligned slot pins which you have to push back into place before inserting a game
- Other physical issues with slots fitting in the cartridge slot

Conversely, are there any people with non-working Everdrives who do *not* experience any of the above issues with **official** carts?

tkbryant, 2021-12-31:
Neither of my Everdrives wiggle at all.

sssembler, 2021-12-31:
Just to add to this: my everdrives work great and have done since I got the Pocket 2 weeks ago or so.   No issues with the socket either - carts are absolute solid once in and no amount of shaking or tapping will budge them.

bender-the-greatest, 2021-12-31:
@tkbryant Sorry, I missed a bullet point on that check list. Do you have any issues at all running official games? This includes having to clean your cart pins more often than you would on a real GBA for the game to be recognized.

mchiaramonte, 2021-12-31:
I did some testing today with my GB X3 cartridge running with the force to Gameboy turned off and it’s definitely more stable that way. I had been forcing Gameboy for that cool green screen effect but though it was worth a shot. With it in GBC mode, although I did run into one problem at first, I was able to use it pretty much without issue. I tried Super Mario Land 2 which had been giving me constant problems and things seemed fine there. This is on firmware 1.0, haven’t tried it on firmware 1.0a that way, but I’ll have to give it a shot.

tkbryant, 2021-12-31:
No unfortunately I sold off my collection of carts between the last couple
of years, I’ve only got the Everdrives now.

coolbho3k, 2021-12-31:
I do have a feeling that the issue is that some cartridge slots are just more finicky than others and the Everdrive PCB/rear cartridge shell might be particularly thin. The tolerances work out on real hardware but cause issues on some Pocket units.   

I don’t have the equipment, but if someone has good calipers, do you mind doing some measurements of the total width of the cartridge shell, and the width of the back half + PCB, on both an Everdrive X series and a real game?

bbbbbr, 2021-12-31:
> I don’t have the equipment, but if someone has good calipers, do you mind doing some measurements of the total thickness of the cartridge shell, and the thickness of the back half + PCB, on both an Everdrive X series and a real game?

Keep in mind it's a small anecdotal measurement

EverDrive GB X3 (SD errors some of the time):
- Total thickness of shell: 7.4mm
- Thickness of back half + PCB: 2.15mm
- Thickness of top of shell: 1.4mm
- Inside thickness from PCB to top of shell: 3.85mm

Two commercial GBC games:
- Total thickness of shell: 7.5mm
- Thickness of back half + PCB: 2.30mm
- Thickness of top of shell: 1.2mm, 1.3mm
- Inside thickness from PCB to top of shell: 4.0mm, 3.9mm

coolbho3k, 2021-12-31:
From the one measurement it sounds to me like the pins on the Everdrives are 0.15mm recessed compared to commercial games.

mchiaramonte, 2022-01-01:
So I disassembled my everdrives and an original GB and GBA cartridge and you’ll notice that the everdrive GB X3 pins don’t come up quite as high as the GB vs the GBA X5 mini. I wonder if this is causing the issue. If I shift my x3 up a little bit repeatable glitches that I’ve found will go away.  

[image]

[image]

Ka0tiK, 2022-01-02:
Hi folks.
I have two GBx7 and two pockets - both generate the IO read error. There are rare cases where I can get a single game to load if it had pre-loaded before on OG hardware; otherwise, the cart is useless. I should mention that I bought both GBx7 genuinely and a year apart from each other. Both pockets boot OG games without issue.

coolbho3k, 2022-01-02:
> Hi folks.
> 
> I have two GBx7 and two Pockets - both generate the IO read error. There are rare cases where I can get a single game to load if it had pre-loaded before on OG hardware; otherwise, both carts are useless for the Pocket. I should mention that I bought both GBx7 genuinely and a year apart from each other. Both Pockets boot OG games without issue.

Good to know. Just curious:

1. You can only load the single game rarely? What happens in most cases if you press "start" on the X7 to load a previously loaded game?

2. When you manage to load a game, do you ever notice any glitches? Ie. graphical corruption and crashing.

Ka0tiK, 2022-01-02:
> 

1. I can elaborate on two conditions: 

2. sdf

Ka0tiK, 2022-01-02:
> Good to know. Just curious:
>
> 1. You can only load the single game rarely? What happens in most cases if you press "start" on the X7 to load a previously loaded game?
> 2. When you manage to load a [...]

I can elaborate on the conditions I have tested:

1. If I load (i.e. pre-load) the game on OG hardware, I am able to boot the game on the Pocket using the START button. I will stress that I have limited experience using this method, but the few times I have tried this, it has worked. Understandably, my experience is limited as it is a cumbersome process just to play a game on the Pocket, especially when I have OG hardware I can use.
If I start the GBx7 on the Pocket, I have rarely been able to successfully start the game (without using the pre-loaded method mentioned above). In short, the very first boot up on both GBx7 I was able to select any game; I know this as I had selected new games that were not pre-loaded before. However, beyond this first load, I have never been able to select another game successfully. There have been one or two cases where I was able to switch to another never-played-before game, but since then both GBx7 have been useless.
I've seen some folks mention about "Select Only" or other types of methods to store the game in recent history and then START as a work around does not work for me. If I start a game and get the IO error, even START does not bypass this error.
2. I have never played the games long enough to notice any glitches. TBH, the frustration before I am able to get a game to load on the Pocket makes the situation not worth playing more than a couple minutes just to admire the screen. :) If there is interest for a long play performance, I am willing to give it a try.

coolbho3k, 2022-01-03:
@Ka0tiK thanks for the tests... unfortunately since neither of your X7s nor Pockets work we cannot deduce whether it's the Pocket or the X7 is causing the issue for you :( 

However, the act that games seem to not glitch out for you is slightly promising, as it means most or all of the pins must be making good contact.

coolbho3k, 2022-01-03:
Another thing to test is, if anyone has a quality multimeter: if you take the front half of the X7's shell off and insert it into the Pocket, you should have enough room to able to test the continuity and resistance between the Pocket's cartridge slot and the flashcart. Perhaps all the pins are making contact but in some cases there's too much resistance on some of the pins. 

If someone has both a working and a non-working X7, this may be something worth looking into.

Not sure though, I'm sure Kevtris has been working on this problem...

KeenIIDX, 2022-07-30:
Today’s 1.1 beta firmware did not make my X7 work.

coolbho3k, 2022-07-30:
It seems like it might be not fixable in software to be honest.