frederic-mahe/Analogue-Pocket #5: 8bitdo SN30 2.4ghz disconnect / reconnect dock issue

dereckhall, 2021-12-26:
I have 6 different 8bitdo SN30 2.4ghz controllers. I only pair 1 to the Pocket Dock fw 1.0
It works fine for a bit, but I constantly receive disconnected/connected in the bottom left corner. There have been a handful of times where the controller has disconnected entirely and the only way to get it back to a working state is to unplug power from the dock and it'll automatically repair and start working again.

I have duplicated this behavior with all of my SN30's and the Pocket dock. I do not have this issue with the same controllers and my other Analogue consoles.

jduckett95, 2021-12-26:
Worth remembering the only 2.4G controller officially supported by Analogue OS 1.0 is the 8BitDo Arcade Stick. The SN30 2.4G will be officially supported from Analogue OS 1.1. As it pairs fine now, I imagine this very issue is why it isn't listed as being officially supported yet.

BestPig, 2021-12-27:
I had the exact same issue with a Switch Pro controller that is officially supported.
Also had to unplug power to force it to restart, undocking/ docking the Pocket didn't worked.

spoonmanx, 2021-12-27:
How about the other two main 8Bitdo 2.4g controllers, the M30 (Gen), and N30 (NES)? Will those be officially supported in the 1.1 Dock update? Also, will two of each be supported, or is it two TOTAL 2.4g controllers, maximum?

velipso, 2021-12-27:
@spoonmanx You can see what Analogue plans to support in v1.1 in the Dock manual, here:

https://assets.analogue.co/pdf/9171a59efe42e5378e2b8489c3cbd19e/Analogue_Dock_User_Manual_v1.1.pdf

If you scroll down, there is a grayed out section for firmware v1.1.

dereckhall, 2021-12-30:
1.0A firmware for the Pocket was released today
https://www.analogue.co/support/pocket/firmware/1.0A

It contains several fixes, but one notable
"System: Rare controller / dpad stops working issue"