frederic-mahe/Analogue-Pocket #10: Altered Beast: Guardian of the Realms does not load save

breiher, 2021-12-27:
Played with legit cart. Game has an autosave feature. Turning off or reseting game on Pocket results in "New Game" screen that otherwise only results on first boot of the game. Turning off game on Pocket and booting on original hardware results in save being loaded as normal. Tested with flash cart, save file is created but not read.

breiher, 2021-12-29:
They advertising this game on the Pocket twitter account even tho it doesn't run on the hardware, hope that means a fix is coming.

GreatHierophant, 2022-01-06:
A workaround for this issue is to start a new game and play until you beat the first level on an original GBA or in an emulator with the official GBA BIOS loaded like mGBA.  After that then you can play, save and load a game without further issue on Pocket.

GreatHierophant, 2022-01-11:
Further testing shows that this game may be buggy and have difficulty saving at times.  One method that appears to give a reliable save on any device is to start a new game, beat level one, quit the game, go to review progress and and press select to be given the option to delete your save game data.  Delete the data and that will prepare the EEPROM data properly and you should be good to go start fresh.

francoiswnel, 2022-08-02:
@frederic-mahe According to the release notes, this is not a Pocket bug. Should be okay to close. https://www.analogue.co/support/pocket/firmware/1.1-beta

"Altered Beast: Guardian of the Realms: save loading is broken in the game itself and this is not a pocket bug; on the everdrive and friends you must clear the save data yourself: select "review progress" from the main menu, then press select to reset the game data. YES to confirm."

closed 2022-08-03 by frederic-mahe