frederic-mahe/Analogue-Pocket #11: Buttons on the Pocket will sometimes become unresponsive after booting a game

OngoGablogian, 2021-12-27:
I've noticed that every now and again after booting a game (OEM Cart, Flashcart, or .pocket GB Studio game) the consoles buttons will become unresponsive until I power cycle the Pocket.  When this happens the home menu works as do all the buttons while in the Analogue OS, but nothing registers in game.

zypherion, 2021-12-27:
I've seen this as well, on a legit Pokemon Pinball cart and on my Everdrive GB (Pre-x series).

16bitstreamer, 2021-12-27:
I have experienced this issue as well. Requires a hard power cycle. When returning to the Analogue Pocket menu the directional pad works but none of the other buttons work.

italiangrandma, 2021-12-28:
Can confirm I've experienced this issue a few times as well.

OngoGablogian, 2021-12-30:
This issue was fixed as of Pocket firmware v1.0A.

coolbho3k, 2021-12-30:
Awesome. Let's mark issue as resolved.

closed 2021-12-30 by OngoGablogian