frederic-mahe/Analogue-Pocket #6: Analogue Pocket Dock HDMI Port Always Powered

TheChrisGlass, 2021-12-27:
**Summary:**
The Dock HDMI port is always powered on when plugged in. There is no sleep mode, regardless of display mode.

**Explanation:**
This is an issue because of the prevalence of Auto HDMI Switches. The "Auto-switching" HDMI docks use the power-on signal of an HDMI cable to know when to switch to the input. Because of their simplistic logic, each HDMI Switch only can support one "Always On" HDMI device at a time. The Xbox One/Series is another device that has an "Always On" signal state. That means you can't properly use the auto-switching on an HDMI Switch if you both an Xbox One and an Analogue Dock plugged in.

While I would love Microsoft to **also** fix this issue, I have to imagine I will have more luck with Analogue fixing it first.

The only workaround right now is to physically unplug the HDMI cable or power off the Dock when not in use.

**Currently:**
HDMI port always sends a signal when it has power.

**Expected:**
HDMI port is unpowered when not in use.

ifighftdragons, 2022-01-19:
I'm experiencing the exact same issue. Hopefully a firmware update can fix this.

ifighftdragons, 2022-11-28:
I can't believe this still hasn't been adressed by Analogue.

iPruch, 2022-12-28:
Same issue here. My TV always shows the Pocket’s HDMI as ON. Also, the green LED glowing always make me think that I turned the Pocket ON by accident, but it’s not…