frederic-mahe/Analogue-Pocket #7: Sleep mode causes save file to be erased Pokemon leaf green

Xukkorz, 2021-12-27:
Using sleep mode on intro cinematic can cause save file for Pokemon leaf green to be erased. Doesn't happen every time but was fairy quick to replicate after a few sleep wake cycles.

Closing and reloading does not cause save to show back up. 

Game used was Japanese version of game and is original.

NadeEater, 2021-12-27:
Reporting the same issue on the French version of the game, original cartridge as well. 

However quitting the game then restarting will generally bring back the save file though success isn't guaranteed, furthermore creating a save state on the title screen will also erase the save file, quitting then restarting the game will restore it.

4wilko, 2022-01-01:
I'm not sure if this should be posted as its own issue, but I can consistently wipe the save data in Kirby's Pinball Land with sleep mode as well under the release 0.9 firmware along with the currently newest 1.0A firmware. Sleep -> Wake -> Close game -> Play cartridge would lose the high scores every time. Fully powering off the Pocket (with or without removing) or normally closing and starting the game retains the high scores just fine. I did not test sleep mode with the GBA games I tried and I'm pretty glad I didn't now.

4wilko, 2022-02-19:
Updated to the 1.0B since I saw it had the Wario Land save/wake fix listed. I no longer seem to be able to replicate the issue on my Kirby's Pinball Land cart so I think the problem is fixed. Can anyone else confirm?