frederic-mahe/Analogue-Pocket #8: Golden Sun: The Lost Age menus are very buggy

bender-the-greatest, 2021-12-27:
When importing the transfer password, the icons for character selection and current position are misplaced. Note that if you input the password perfectly the first time you won't have to worry about not knowing your backwards position, but especially with Gold passwords this is incredibly tedious and frustrating to encounter when you *do* get part of the password wrong. 

Also, Djinni are completely unmanageable, aside from being able to use and set them in battle. The overworld Djinni menu is bugged and will select unused rows resulting in occasionally buggy behavior on selection (like being able to set Flint on a party member at the start of the game despite not having it yet). Moving the cursor enough results in the selection moving offscreen and crashing the game. Glitchy behavior aside, it makes it impossible to move Djinni between party members.

I have confirmed neither of these things occur on a real GBA.

tjanas, 2021-12-27:
@bender-the-greatest Is this with a real Golden Sun: The Lost Age cartridge, or via a flash cart?

bender-the-greatest, 2021-12-27:
This happens with a real cart. I can try to get a gameplay video uploaded today. To be clear I did test this on two AGB-001s which work as expected.

bender-the-greatest, 2021-12-27:
@tjanas Here are videos of both. Each video includes an unmodified (aside from the shell) example of the same game areas working correctly:

[Analogue Pocket Firmware 1.0 Golden Sun: The Lost Age Djinn menu glitch
](https://www.youtube.com/watch?v=yq6KAr5SVd4)
[Analogue Pocket 1.0 Firmware Golden Sun: The Lost Age transfer menu glitched
](https://www.youtube.com/watch?v=Q1rSiOo4doM)

Curiously, the first Golden Sun has no notable issues (only noticed one minor animation not playing) I noticed playing through the game on the Pocket, despite the game engines being very similar.

bender-the-greatest, 2021-12-27:
Note that while the game is playable with both issues present, anyone familiar with either game knows that Djinni management is critical to class building and building out your teams effectively and that the correctly functioning password is already tedious enough on its own. The game is technically playable, but these are major issues that will major functionality of the game until fixed.

coolbho3k, 2021-12-27:
+1.

I can reproduce with my EZ Flash Omega DE.

GreatHierophant, 2021-12-27:
Can you post a save file for the Djinni issue?  The password screen issue can be observed almost immediately upon booting the game, but I don't know how far you have to get into the game to be able to access the Djinni menus in a way that will show off the bug,

bender-the-greatest, 2021-12-28:
Unfortunately I don't have a way to dump the save, unless there is some way with the Pocket I'm not aware of. This can be reproduced after the beginning of the game once you receive the first Djinn outside of Daila (the village you land in after Idejima crashes). This should happen within 30 minutes (if that) of gameplay after starting a new game.

GreatHierophant, 2021-12-28:
I do not have the original cart, but can reproduce the two bugs mentioned with my EverDrive GBA X5.  I also have found that the regular menu, when I played the game as far as the first battles, is very glitchy.  You can put the cursor on blank spots on the screen, see characters which aren't in your party and it will crash if you move through the menus enough times.

tkbryant, 2021-12-29:
Yeah it hangs up for me in the item menu as well, buttons don't work, it just semi crashes, you must quit the game to get out of it.

tjanas, 2021-12-30:
Is this still broken in firmware update 1.0A?

bender-the-greatest, 2021-12-30:
I'm not sure, but I can update the firmware when I get home in a couple of hours.

coolbho3k, 2021-12-30:
1.0A did not fix the password entry glitch for me.

tkbryant, 2021-12-30:
1.0a does nothing for the issue,does anyone else's game lock up on the "Status" screen when choosing that option?

bender-the-greatest, 2021-12-30:
@tkbryant I hadn't noticed the Status screen issue on 1.0 but others had reported this on a reddit thread I started about this problem. That said I just tried it on 1.0A and the Status screen is bugged out, but at least I can navigate away from the garbage entries (selecting offscreen causes graphical artifacts and can trigger a crash same as the Djinni menu). The Djinni management menu does not let me move the cursor up, however, and remains largely unusable.

@tjanas As others have chimed in, 1.0A does not fix either issue outlined in my OP, nor the aforementioned Status menu bug.

tkbryant, 2021-12-30:
Yeah it was an issue in 1.0 for me too but I just hadn't seen anyone
mention it based on what I had seen at the time, glad to know it's an issue
for everyone & can be patched up hopefully.

coolbho3k, 2022-01-01:
For some reason I can't reproduce the aforementioned Djinn bug in The Lost Age on my Omega DE on 1.0A. The Djinn menu seems to work normally for me. That's really weird that it isn't showing up for me.

I still get the password bug every time I try, but that means the game SEEMS to be playable for me.

tkbryant, 2022-01-01:
What about the “Status” screen? It freezes my game every time I select it.

bender-the-greatest, 2022-01-05:
FYI I tried a new game with no import to see if there was any change. Didn't even get out of Venus Lighthouse, the status screen glitched the F out on entry.

bender-the-greatest, 2022-01-24:
I received an update from Analogue support this morning regarding **Golden Sun: The Lost Age**. Short answer is that these issues should be addressed in the next firmware update. Here is the full response (name redacted) from Analogue:

```
Hi REDACTED,

Apologies for the late reply. We really appreciate this kind of detailed feedback and thank you for your patience. This is a known issue and will be fixed on the next firmware update.

Have a great day!

Best Regards,
Analogue Support
www.analogue.co
```

To be clear, this is in response on reporting to them the Status menu crashes, Djinni menu crashes, and Password screen difficulties in great detail. Hopefully the next firmware drops soon so we can confirm.

bender-the-greatest, 2022-02-01:
All of the issues reported in this thread are now fixed with firmware update 1.0B.

closed 2022-02-01 by bender-the-greatest