frederic-mahe/Analogue-Pocket #15: DKC Roms will not save.

SSj-CODii, 2021-12-28:
I am using an EZFlash Omega updated to v9. The Donkey Kong Country trilogy of games result in a “save failed” on every attempted save. The roms otherwise work flawlessly.  Other tested roms with different save types save and load as expected

coolbho3k, 2021-12-28:
Sounds like an EEPROM save issue on this flash cart just like in the GBA X5 in #2. Note this is the original EZ Flash Omega, not the EZ Flash Omega Definitive Edition.

SSj-CODii, 2021-12-28:
I got reports from another user that Boktai and FFVI are also not working. Are they EEPROM as well? And yes you are correct this is not the DE I am using. I’ve seen at least one person comment the games are saving correctly on DE

coolbho3k, 2021-12-28:
@SSj-CODii can you edit the title to include the fact that you're using an EZ Flash Omega?

gutxihaitz, 2022-01-09:
I have a similar problem. I was playing Altered Beast - Guardian of the Realm (European ROM) using EZ-Flash Omega on my original GBA. I switch to the Analogue Pocket and it looks like if there is no save data. On the GBA there is no such problem.

SSj-CODii, 2022-01-10:
> 

The issue seems to be with EEPROM save types. I have been able to get the games saving on the Analogue Pocket by using GABATA to patch the save type of the rom to SRAM. It’s the best workaround for now.