frederic-mahe/Analogue-Pocket #17: Pocket unit "crackles" when disconnecting from Dock

filler-arc, 2021-12-28:
When the Pocket is removed the Dock, it emits an audible "crackle" sound from the USB C port.
This happens with the Dock connected and disconnected from its power supply.

I attempted to reproduce the noise using other cables/chargers (iPad charger, Anker power brick, etc.), but the noise is only produced when disconnecting from the Dock.

schmingy, 2021-12-28:
We ordered his and hers docks and pockets. One dock we have exhibits this problem with both pockets and the other one does not. It almost sounds like pins having been deformed springing back into place.

bbbbbr, 2022-01-04:
A reply from Analogue is quoted here:
https://www.resetera.com/threads/analogue-pocket-ot-hey-kid-wanna-score-some-tetris.147414/page-123#post-79032874

> Analogue responded to me about the crackling issues and this is what they said
> > There is no need to worry as this is normal. The USB-C connector contacts are making this sound and as the Dock is used more, the sound will diminish.
> >
> > Should there be any further queries or concerns, please do not hesitate to contact us again.
> > 
> > Thank you!

SpiralBrad, 2022-01-08:
Unplug the pocket from the dock in the dark, and you will see that there are actual sparks / arcing involved.

I'm not sure how this is 'normal' or how it will reduce over time...

schmingy, 2022-01-08:
In the case of ours, the sound is present even with the dock unplugged for power and hdmi. The only light I’ve noticed is the led on the pocket lighting up upon removal. That’s not to say that some are seeing sparks, just not what we’ve seen with ours.

SpiralBrad, 2022-01-09:
I captured the sparking on video as best I could: [https://imgur.com/a/A9KrueL](https://imgur.com/a/A9KrueL)

The camera is on a tripod, so once it goes black it remains pointed right at the port area of the dock. With the removal at ~0min 14sec, you can hear and see sparking.  It's not easy to capture (or rather, I don't know how is best) but... it's definitely not just "connector contacts" making this sound.

...It shouldn't be making LIGHT in pitch darkness, I'd say.

16bitstreamer, 2022-01-09:
Is the light not the led of the bottom of the Pocket coming on for an instant as the USB disconnects? 

Try connecting the power cable to the Pocket directly and disconnecting while it’s plugged in to see if you can replicate the light?

schmingy, 2022-01-09:
I just replicated the light issue using just a USB C cable. The light is the power LED on the Pocket that briefly goes from yellow to red after the USB C cable is unplugged. This does not appear to be a spark despite what it sounds like. 

Testing with a bog standard USB C cable and power supply, I noticed the same effect of light for a very short duration after unplugging. Trying the same thing with an unpowered dock, and while I get the sound, no light. When I tried with a powered dock, same sound and I do see light; however, looking closely, the light comes from the same place as the power LED and not where the USB C connector is. Finally, the color of the light doesn't seem right for what a spark would look like. I believe a spark would most likely be blue/white in color.

When the Pocket charges, the LED on the bottom is lit, when unplugging, there appears to be a delay in turning off and it goes from orange (at least on my not fully charged pocket) to red before finally shutting off.

SpiralBrad, 2022-01-09:
> When the Pocket charges, the LED on the bottom is lit...

I uhh... I forgot there were LEDs on the bottom.  I'm an idiot :)

But phew that does make much more sense.

schmingy, 2022-01-09:
> I uhh... I forgot there were LEDs on the bottom. I'm an idiot :)
> 
> But phew that does make much more sense.

Please don't be so hard on yourself. If it's just the LEDs then at least we maybe aren't dealing with a sparking/electrical issue, but that you were asking and looking into it means that you wanted to learn. That means you're not an idiot. :)

The sound is concerning, but, to me at least, it sounds a lot like little pins popping back into place when my ear is close. That being said, I don't know why some docks have that sound and others don't unless they just have strange tolerances. The USB C ports on the dock are mounted in something flexible so they can move around a little bit as a relief strain. 

If something really is flawed in the design, I'm sure Analogue would address it as they did with the defective Game Gear adapters they sold for the Mega SG.