frederic-mahe/Analogue-Pocket #18: Pokemon Puzzle Challenge for GBC - save data is wiped upon being inserted into Analogue Pocket

better0nholiday, 2021-12-28:
So I recently got a copy of Pokemon Puzzle Challenge with a dead battery. I soldered in a replacement and put the cartridge into my Pocket and it played fine. I put a couple of hours of playtime into it over a few sessions when I noticed my save data was gone. I just assumed the battery was a dud and soldered in another one. I tested the game in the Pocket again and same thing; upon booting the game up again after creating a new save, the data gets wiped. I then did a bunch of tests.

-I put the cartridge into a GBA SP and created a new save. Upon several power cycles, the save remained intact. I swapped the cartridge between 2 other GBA SP's and a GBC and the save still remained intact. I then placed the cartridge back into the Pocket and upon booting it the save was wiped. I repeated several times and always got the same results.

-I created a new save (on the Pocket), tried entering and exiting sleep mode, and the save remained intact. Exiting the the main Pocket menu and re-booting the game however causes the save file to be wiped.

-I created a new save (on a GBA SP), placed the cartridge into the Pocket then took it out WITHOUT booting it. I placed the cartridge back into a GBA SP to find the save data had been wiped.

-I created a new save (on the Pocket) and pulled the cartridge out without closing the game first (I know that's a big no-no, but I was curious). I then put the cartridge into a GBA SP and... the save remained intact.

As best I can figure there is something weird going on with the Pocket's main menu/OS causing the save to be wiped. I posted this issue on Reddit and so far no one else has been able to replicate this problem with their copy of the same game. This leads me to think it's maybe an issue with the replacement battery, but I have tested several other games in the Pocket which used the same replacement batteries and those all seem to work fine/saves remain intact.

I would try yet another battery, but I have to wait until I get more in as I used my last 2 on this cartridge.

LoneBoco, 2021-12-28:
I noticed my Zelda DX save got deleted and I assumed it was just a dead battery.  I'll keep an eye out to see if it gets deleted again.

16bitstreamer, 2021-12-28:
I experienced this error with Dragon Warrior 1&2 of my Everdrive GB x7. I was able to restore the save from one I had backed up. Haven’t been able to replicate the error, however, saw people on Reddit hypothesize that it has something to do with the sleep function, but sleep doesn’t work with the Everdrive, so, I’m not so sure.

better0nholiday, 2022-01-03:
Alright so some minor updates. On a whim, I decided to open the cartridge up and use some 91% alcohol on a q-tip and clean around the battery contacts. There was a little minor residue that got picket up. This fixed it! Or so I thought. The save would no longer INSTANTLY be wiped upon boot, but rather if I left the cartridge in the Pocket for more than a couple minutes on the Pocket's main menu, next time I started the game up it WOULD wipe the data. I was able to replicate this a number of times. So now I'm un-sure if the cleaning actually accomplished anything or not.

Today I FINALLY got some fresh batteries in. I did an extra thorough job of getting all the old solder off the battery pads, used new solder, new flux etc... same results. If I just pop the cartridge in and boot it right away, it usually (but not always) maintains the save, but if I leave the cartridge in for a few minutes while the Pocket is on the main OS menu, it's guaranteed to wipe the data.

I also noticed something. Whenever I boot the game and it DOES wipe the data, there is a split second white flash before the copyright screen (the blue screen with Pikachu). Anytime it doesn't wipe the data, there is no white flash.

So at this point I'm not sure if my copy of the game has a problem (visually inspecting it I can't find anything possibly wrong with the board, and it still functions perfectly in any other Gameboy console) if there is a weird compatibility issue with the Pocket or some strange combination of the two. I'm tempted to order another copy of the game but if this is a issue that can be fixed via firmware down the line, I'd rather hold off.

neutralinsomniac, 2022-01-26:
I have a repro cart of Megaman Battle Network 3 (GBA) that has had its save file wiped twice on the Pocket. I'm assuming it's something the Pocket itself is doing since I've had zero issues with the cart on real hardware.