frederic-mahe/Analogue-Pocket #19: Wario Land 3 crashed when entering the golf minigame

bramblescram, 2021-12-28:
The screen turns white and the game crashes. I have reproduced the error three times.

TheMostSecret, 2021-12-29:
Haven't tested out the golf mini game yet, I will try it later tonight and will edit with if my game crashes too. 
But an issue I've had with WL3 is that sleep mode doesn't work with it. Upon waking the game will have visual glitches and Wario will fall through the floor.

I'll try and test out this game more later and see if I can find more bugs.

GreatHierophant, 2021-12-29:
I cannot reproduce the error with an EverDrive GB X7, the golf game works fine for me.

BestPig, 2021-12-29:
Probably playing from my converted game as a pocket file.
I will try to fix my patch, but this is not related to a Analogue Pocket bug.

PrimoAngelo, 2021-12-29:
Fo me, only converted game to .pocket freeze with white screen entering all the golf minigame doors. Original card or rom+flashcard is working fine. But as mentioned by TheMostSecret, sleep mode is bugged with the game even with original cartridge.

tjanas, 2021-12-30:
@bramblescram If this is only a bug with a converted .pocket file, please change the title to reflect that it’s a bug with the sleep functionality in the game.