frederic-mahe/Analogue-Pocket #20: GB Studio games not saving in "GB mode"

TheMostSecret, 2021-12-29:
Title say it all really, when you run a GB Studio game under GB mode the game doesn't save. 
This can be best shown with 'El Dueloroso' which only runs in GB mode. When playing the game high scores are not saved. I thought it might be a glitch with the game but testing out other games (and turning on 'Force GB mode' as other released compatible games run in GBC mode) shows that it can load save files just fine but it can't make new ones in GB mode.

16bitstreamer, 2021-12-29:
Can confirm across several .pocket games I’ve tried, including a test built from examples in GB Studio.

OngoGablogian, 2021-12-30:
This issue was fixed as of Pocket firmware v1.0A.

tjanas, 2021-12-30:
@TheMostSecret please re-test with v1.0A and close the issue if you can confirm it is now resolved.

TheMostSecret, 2021-12-30:
Looks like things are saving now!

Closing issue...

closed 2021-12-30 by TheMostSecret