frederic-mahe/Analogue-Pocket #21: Color space in GBA original LCD mode for is inaccurate

coolbho3k, 2021-12-29:
The Pocket appears to currently implement a trivial desaturation filter for the original LCD modes for GBA. In reality, this is not how the GBA display actually looked.

The community has done deeper research into replicating the intended look of the original GBA display. Long story short, for the most accurate colors, the Pocket needs to implement the proper LUTs to transform the video output into the correct color space.

See Pokefan's post on this for the technical details: https://forums.libretro.com/t/real-gba-and-ds-phat-colors/1540/159

Peredonov, 2021-12-29:
It is important to note that this issue is also relevant for GBC.

Another source of technical information by Extrems: https://www.gc-forever.com/wiki/index.php?title=Game_Boy_Interface/Standard_Edition#Color_matrix

16bitstreamer, 2021-12-29:
Does anyone have a screen color calibration test that can optically confirm? (I’m talking about something like I would use to calibrate my monitor to match my photography)

bbbbbr, 2022-01-04:
This isn't a calibrated test exactly, but you can use this image + rom pair. You could run them side-by-side on a GBC and AP

[color_wheel.zip](https://github.com/frederic-mahe/Analogue-Pocket/files/7804791/color_wheel.zip)

coolbho3k, 2023-05-04:
Is this fixed in the 1.1 update released today?