frederic-mahe/Analogue-Pocket #23: Legend of the River King (GB) - Game Freeze?

199X, 2021-12-30:
Game will boot up but when you press Start at the intro screen it fades to white and never continues to the game selection menu. I verified that the game was working on a GB and GBP. Not sure if anyone else has had this issue or could test their cart.

coolbho3k, 2021-12-30:
I cannot reproduce this bug with my ED GB X7 (but that doesn't mean it doesn't have issues with the real cart).

tjanas, 2021-12-30:
> Game will boot up but when you press Start at the intro screen it fades to white and never continues to the game selection menu. I verified that the game was working on a GB and GBP. Not sure if anyone else has had this issue or could test their cart.

Can you share a video?

199X, 2021-12-30:
> > Game will boot up but when you press Start at the intro screen it fades to white and never continues to the game selection menu. I verified that the game was working on a GB and GBP. Not sure if anyone else has had this issue or could test their cart.
> 
> Can you share a video?

I'll post a video tonight.

BestPig, 2021-12-30:
Are you playing with a real cartridge and not a flashcard?
Because even when some seems to works fine, SD I/O are unreliable and make some glitch or crash sometime.

199X, 2021-12-30:
Ahh yeah... forgot to mention that. It's an original cartridge.

199X, 2021-12-31:
Here's a video: https://youtu.be/fb26QLlyA98

I did see that there's a new firmware version so I'll download that later and see if that changes anything.