frederic-mahe/Analogue-Pocket #25: Dock issues (wireless connectivity, and game "freezing")

SamuraiSamH, 2021-12-30:
1. When I first used my dock I was able to sync it with an 8bitdo wireless controller. Then one time I put the system in sleep mode and couldn't get the controller to reconnect, and haven't gotten it to connect to a Bluetooth controller since. Pressing the sync button on the dock doesn't seem to do anything, the system does not appear to go into pairing mode as the lights do not move nor does the system connect to a controller. In the meantime I've been connecting controllers via their USB cables.
2. Sometimes when playing on the dock, the system does a weird kind of "freeze" where it no longer detects any input from my controller nor will it reconnect to whatever controller is connected. The game itself will continue playing, and if I was holding down an input at the time of the "freeze" it will assume that button is still being pressed. I am able to take the Pocket out of the dock and resume play as normal when this happens. If I do that, the game still remains on screen running normally as if I had left the Pocket on the dock. The only way for me to "unfreeze" the dock is for me to unplug it. This has happened with both Game Boy Color and Game Boy Advance titles.

tkbryant, 2021-12-31:
Yeah I'm having the exact same issues, it's completely random & can happen on any game.

frederic-mahe, 2022-01-01:
could be similar to #5

mddifilippo89, 2022-02-03:
@frederic-mahe 
How important is to use a supported controller at this time? 
The reason I ask is because I upgraded from 1.0 to 1.0B and now the dock no longer wakes up the pocket with a bluetooth controller that I had working consistently prior during. 

The snes switch online controller

frederic-mahe, 2022-02-03:
@mddifilippo89 I would say that issues with common controllers should all be reported. The switch snes bluetooth controller might not be officially supported, but it is common and informing other _Pocket_ users is important.

I think (I am not an Analogue employee) that testing outside the initial range of use and reporting many use cases will help improve the product, and at the very least, its documentation.