frederic-mahe/Analogue-Pocket #26: Problems with Pokémon Yellow

bendikredick, 2021-12-31:
I'm having trouble launching Pokémon Yellow on the Analogue Pocket. Most times I get an error message. I have tried with several original copies of the game, and they all showcase the same issue. On the few occations the game actually starts, it runs in GBC mode, even though it is a GB game (with some enhancements when running on a GBC). I suspect this might be part of the problem, that the Pocket doesn't know if it should launch in GB or GBC mode? Anyways, the games all run smoothly on my og GBC.

GreatHierophant, 2021-12-31:
Pokemon Yellow is, in its code, a hybrid Game Boy/Game Boy Color game, so it should by default use the GBC mode.

Quentin01, 2021-12-31:
My two cartridges of Pokemon Yellow (french version) have no issue starting and playing. I cleaned up their pins before trying them maybe that's your issue?

And yes, as it is an "hybrid" game, to force the GB mode you can do it from the settings of Analogue OS.

tjanas, 2021-12-31:
> I'm having trouble launching Pokémon Yellow on the Analogue Pocket. Most times I get an error message. I have tried with several original copies of the game, and they all showcase the same issue. On [...]

Can you share a screenshot of the error?

coolbho3k, 2021-12-31:
Have you tried any other original carts?