frederic-mahe/Analogue-Pocket #32: GB plays as GBC - F-15 Strike Eagle

danhood, 2022-01-02:
GB game loads in GBC mode.

OS Version: `v1.0a`

Cart Code: `DMG-EG-USA`

Original Cartridge: `YES`

Dumped ROM's SMDB Entry:

```
571e45610cae528b3c50304c42c946f307d71797fc2029e9ca1978045dd5186d	Game Boy/1 GB US - A-G/F-15 Strike Eagle (USA, Europe).gb	237a08abe860bce316f1ce337afb11ee0e566037	a8b74d14e66f7a3049e79ca147141d52	045dee8c
```

sssembler, 2022-01-04:
Force GB mode - this is how it works for the moment.

coolbho3k, 2022-01-05:
Ok I figured out why the Pocket is doing this with only some GB games.

The way a GBC game is detected is the upper bit of byte 0x143 is set to `1`. 

On newer games this byte is reserved for flagging whether the game is a GBC game.

On some older Game Boy games the byte wasn't reserved and game's title can overflow into byte 0x143. In the case of F-15 Strike Eagle, this byte is set to 0x45, the last `E` in `F-15 STRIKE EAGLE`, and the upper bit of 0x45 is set.