frederic-mahe/Analogue-Pocket #33: EZ Flash Omega - NOR flash games lead to white screen

jibblerdahut, 2022-01-03:
All games seemingly work fine booting off of SD. Even ones that typically have problems, like Mother 3 and Pokemon Fire Red. Both boot fine when launched off the SD card directly.

Writing a game to NOR flash succeeds, but booting the game from NOR flash results in an almost instant white screen. The "Loading..." black screen shows for a split second, then immediately goes to a white screen. 

Have tested this on Mother 3, Fired Red, and Altered Beast. All no-intro. Have swapped SD cards, tried exFAT versus FAT32, fully formatted cards twice, and utilized other backups.

coolbho3k, 2022-01-04:
For what it's worth NOR doesn't work on the Omega Definitive Edition either - see #1. Glad to see there's some consistency at least.

However, I think someone on GBAtemp did say that it worked for their Omega (non-DE).

Something cool to try: I get either get the save state to load and then immediately freeze or some _insane_ behavior if I save state on a game when loaded from SD card, then try to restore during the white screen after booting the same game from NOR.

jibblerdahut, 2022-01-05:
> However, I think someone on GBAtemp did say that it worked for their Omega (non-DE).

Saw this just now too. Not saying I doubt them, but between myself and 4 others on Discord, we all have the same problem. Partially wondering if his cart is more the exception than it is the rule.

coolbho3k, 2022-07-30:
This is reported fixed on 1.1 beta

acejordin, 2023-08-17:
To update this for 2023:

I am having this issue as well. I have tried booting my EZFlash Omega DE in mode B with several ROMs (Mother 3, Pokemon FireRed, Metroid Zero Mission, Mario Kart Super Circuit) written to it on my Analogue Pocket v1.1 and all freeze at a white screen. All appear to be identified by the memories feature (The title, system, developer, region etc are all displayed correctly for each ROM written to NOR) but do not boot. Most of the ROMs freeze at the white screen with no audio, but Metroid Zero Mission plays crackly looping audio while on the white screen. 

I've done the following tests using Analogue Pocket v1.1 firmware, but I did some limited testing using v1.0b that it first arrived with. v1.0b seems to work the same as v1.1 for me. Mode A worked fine in v1.0b while mode B resulted in a white screen with Mother 3. I then updated to v1.1 as it appeared it resolved some peoples issues. But not for me at least.

All testing was done with the EZFlash Omega Definitive Edition on the latest kernel 1.03 and firmware 6.0.

The Omega DE works fine in Mode A, including the ROMs mentioned earlier. 

The Omega DE in mode B works fine with all of these ROMs on original hardware (tested on a GBA SP).

I've ran the Omega DE safe mode 'test card' test which passed for the S98 error ([Step 9 at this gbatemp thread](https://gbatemp.net/threads/basic-repair-guide-of-current-ez-flash-products.601332/)) so there is no indication that anything at all is wrong with the Omega DE itself.

The EZFlash account on gbatemp [implies there is no issue](https://gbatemp.net/threads/ez-flash-product-for-analongue-pocket-current-solution.618133/), which doesn't appear to be accurate, not for everyone at least.

My Omega DE cartridge works perfectly in original hardware, mode A and mode B.

I've tried raising the Omega cartridge out of the Pocket slot a little bit with no luck, though the garbled audio in Zero Mission changed slightly.

I've tried with the Pocket plugged in and charging, and on battery with no difference. I've tried the FORMAT ALL option in the NOR tab of the EZ Flash in Mode A before writing to NOR. I've tried writing the ROM to NOR using the 'WRITE TO NOR CLEAN' option and the 'WRITE TO NOR ADDON' option. I've tried the 'DIRECT BOOT' option in the NOR tab. All result in a frozen white screen.

From what I've gathered it seems that Mode B on the Omega DE and some versions of Pocket v1.1 works fine for some people. The references to it working all appear to be from using 1.1 betas, I haven't read a clear instance with mode B working on final v1.1, but imagine it must be working for some?

I've tried what coolbho3k says above - I inserted a real Metroid Zero Mission cartridge, started it, made a memory/save state, then put in the Omega DE in mode B with Zero Mission in NOR flash, booted it and during the white screen and garbled audio attempted to load the memory/save state. The save state successfully attempted to load but resulted in the insane behavior mentioned above. A fun experiment but no help.

Finally, I also have an EZFlash V RAM Expansion pack cartridge with Mother 3 flashed to it that the Pocket boots and plays just fine.

kylejrp, 2023-08-24:
I can also reproduce the issue exactly the same as @acejordin, I don't have a solution so far.

auyx, 2023-10-12:
I am getting the same issue with my new transparent pocket. I also tried an EZ 3in1 with Metroid Zero Mission flashed to it and get the exact same result.

ulmentflam, 2023-11-25:
I too have the same issue with the new limited edition color pocket. Running into the exact same issue with any game loaded into the NOR.

skyerday, 2023-12-07:
Mode B or Play game in NOR work in AP FW 1.1 beta AKA (BIOS 1.5). Seems failed with white screen after BIOS 1.6.