frederic-mahe/Analogue-Pocket #35: Unable to reset save data with key combination on boot (Pokemon Pinball)

peterjcole, 2022-01-04:
To reset save data on a Pokemon Pinball cart, you have to hold Up, Right, Start and Select while booting the console.

This does not work on the pocket. Presumably other games with similar mechanisms would be affected as well.

manksalot, 2022-02-08:
This issue was addressed in firmware 1.0b: 
> Pokemon Pinball Keypress Blocked - Fixed allowing data to be wiped normally
Confirmed working with an original JP cartridge.

frederic-mahe, 2022-02-08:
@peterjcole please close the issue if firmware 1.0b solves your problem.

peterjcole, 2022-02-08:
I'm not able to test now, as I reset my data on a real GBC and don't want to reset it again. I'm gonna assume since it's in the changelog that it's fixed, so closing for now at least.

closed 2022-02-08 by peterjcole