frederic-mahe/Analogue-Pocket #36: Add ability to force GBC mode on grey cartridge GB games

coolbho3k, 2022-01-06:
There is currently no way to play an original Game Boy (grey) cartridge as if it were running on a GBC, so you miss out on the original LCD modes for GBC on these cartridges. The only way to do this on the Pocket today is to use a flashcart to first boot the Pocket into GBC mode and then launch the grey cartridge game in GBC mode, but this is not possible on a real cartridge.

This is missing functionality from the Pocket that is present in the GBC. It would be important to have this alongside support for custom palettes that is planned. Adding a "Force GBC" toggle as well as palette options to the GBC core for GB games would fix this.

toto99303, 2022-01-06:
+1 on this.

coolbho3k, 2022-01-07:
A _possible_ alternative would be to allow the selection of the GBC LCD filter in GB mode, in addition to supporting custom palettes for GB games. Not sure if this gets us where we want 100% though.

TheMostSecret, 2022-01-08:
I agree. I love all the screen modes but an option to have color palettes with LCD filter would be much appreciated.

coolbho3k, 2022-02-01:
Fixed with 1.0B

closed 2022-02-01 by coolbho3k