frederic-mahe/Analogue-Pocket #38: The Rugrats Movie (GBC) - Flashing HUD and background elements

italiangrandma, 2022-01-07:
See title.

Tested via Everdrive GB. Tested on a real GBA and the issue does not occur.

bvuillemin, 2022-01-07:
Hi! This problem occured on a real cartridge as well. @italiangrandma and I discussed on Reddit about it. https://www.reddit.com/r/AnaloguePocket/comments/ryijcd/problem_with_rugrats_the_movie_on_gameboy_color

JoseJX, 2022-01-29:
This appears to be an issue with Rugrats - Time Travelers as well: https://www.reddit.com/r/AnaloguePocket/comments/sf12es/graphical_glitches_in_rugrats_time_travelers/

ace-dent, 2022-02-11:
I can confirm @JoseJX issue with "Rugrats - Time Travellers". 
Real Cart (DMG-AVMP-EUR). 
Pocket 1.0B. GBC Mode (no issue in DMG mode). 
Flashing HUD (window) element and background graphics. Becomes more broken on 2nd level 'Gold Rush'.

italiangrandma, 2022-07-29:
Fixed as of Pocket firmware v1.1-beta (7/29/22)!

closed 2022-07-29 by italiangrandma