frederic-mahe/Analogue-Pocket #40: LSDj 9.2.L instrument envelopes don't work properly

iamjackg, 2022-01-08:
I'm seeing some odd behaviour with the instrument envelopes: something about the APU is not being accurately replicated. Take a look at this video I shot: I'm moving the cartridge between the Pocket and a GBC, so both consoles should sound exactly the same, but they don't. I also tested the cartridge on a GBP and a DMG, and they sound the same as the GBC, i.e. correct.

https://youtu.be/qb6qdlEZ_3w

katawajojo, 2022-01-16:
Seconding this. This happens when flashed to a cart, or when converted to GB Studio's .pocket format.

bbbbbr, 2022-01-25:
From what I've seen discussed, it's related to LSDJ changing to use "zombie mode" for envelopes which is apparently not fully implemented on the pocket - but that it was being looked into.