frederic-mahe/Analogue-Pocket #41: Infrared port not working or working unreliably

bbbbbr, 2022-01-11:
When testing with a homebrew "learning remote control" ROM (will attach a copy to this issue) on an Everdrive with a Flirc as the receiver, and training it from a standard remote control (sony rm-yd023)
* Works reliably when using a CGB
* Does not seem to work at all running on the Pocket

Also see this thread:
https://old.reddit.com/r/AnaloguePocket/comments/rzz737/has_anyone_successfully_used_the_pockets_infrared/

bbbbbr, 2022-01-11:
(I did not write this homebrew)
[bluecop-homebrew-gbc-remote.gbc.zip](https://github.com/frederic-mahe/Analogue-Pocket/files/7844470/bluecop-homebrew-gbc-remote.gbc.zip)

Press Select + Start to enter and exit training mode

MHzBurglar, 2022-01-11:
I tested two scenarios using Bomberman Max Blue Champion: Receiving IR input on the Pocket from a TV remote control, and sending/receiving IR communication between the Pocket and an original Game Boy Color running Bomberman Max Red Challenger.

**_Receiving_ input from a TV remote control works perfectly**.  It's very reliable and even works off-angle.
This can be tested by playing one of the levels with the clear condition "Step on the tile to go to the Hidden Area", such as level 1-14.  Upon stepping on the tile, the game will ask for input from a remote control.  Upon detecting input, the game will go to a bonus stage.  This stage is accessible from the stage select menu on File 1 of the attached SRAM save file for Bomberman Max Blue Champion.

**Communication between the Pocket and an original Game Boy Color DOES NOT WORK**. Using the Pitch Area in Bomberman Max, I can sometimes get _one_ of the two systems to say "processing" after several tries and re-alignments, but the transfer never completes.  I believe that unlike the bonus stage, which only need to receive IR input, the pitch area requires both systems to both send and receive.  Since receiving input seems to work, **I believe the core issue is that the Pocket's IR transmission not being received by the other Game Boy Color.**  Either the Pocket's IR 'blast' is too weak/dim, or the Pocket isn't timing/coding it correctly so the Game Boy Color is disregarding it.  Another theory is that the power LED is interfering with the IR send; would it be possible to add a Power LED on/off toggle to the system menu just in case?

I'm no expert on this, but I _think_ the 'level transfer' process in Bomberman Max works like this: When the A button is pressed on one of the systems, it initiates a handshake by sending an IR blast to the other Game Boy Color.  Upon receiving this IR signal, the recipient Game Boy sends an acknolwdgement back to the sending Game Boy.  After this, the Game Boy performing the level "send" will send a signal to the receiving Game Boy which tells it to unlock that level in its save file (the level data itself is already on the cartridge but locked.)  I do not know if a confirmation is sent back to the 'sending' Game Boy afterward, but it doesn't seem to me like that would be needed.

This communication issue can be tested using the "pitch area" from the main menu of Bomberman Max to send/receive level data between the 'Blue' and 'Red' versions of the game.  One system must be running "Bomberman Max Blue Champion" and the other must be running "Bomberman Max Red Challenger".  The attached save file for Bomberman Max Blue Champion has levels available to send on File 1, and they can be received by a copy of Bomberman Max Red Challenger regardless of whether or not that game already has save data.

Games used for testing:  "Bomberman Max - Blue Champion (USA).gbc" and "Bomberman Max - Red Challenger (USA).gbc" (no-intro verified dumps)

Save file for testing: 
[Bomberman Max - Blue Champion (USA) SRAM.zip](https://github.com/frederic-mahe/Analogue-Pocket/files/7848619/Bomberman.Max.-.Blue.Champion.USA.SRAM.zip)

coolbho3k, 2022-01-11:
I think you can test whether it's the power LED interfering by just taping over the power LED.

MHzBurglar, 2022-01-12:
> I think you can test whether it's the power LED interfering by just taping over the power LED.

I tried that, but it seemed to make things worse; with the Power LED uncovered, I'm able to (very rarely) get the Game Boy Color to receive a pulse from the Pocket, but the overall communication handshake fails.  When the power LED is covered, it seems to not work at all.  I tried using a smartphone camera to look at the "IR Transceiver" while the power LED was covered to see if I could see a purple light when it's sending, but I couldn't see _anything;_ it was completely dark.  Obviously, I also couldn't see the infrared light through my camera with the power LED uncovered, as the power LED's light would visibly overpower it.  Based on all of this, it's actually possible that the IR transmitter shares a housing with the Power LED and the "IR Transceiver" dot beside is is just the IR receiver portion.  Given their relative placement, it would line up quite well with the placement of the GBC's transmitter and receiver relative to one another in their housing.

That said, the Power LED _shouldn't_ interfere since its light is on a different wavelength from infrared light, but the ability to turn the LED off would be nice to have either way.

When I did the 'smartphone camera' trick on my Game Boym Color, I could see a faint purple light, so the GBS is definitely transmitting something when I press the button.

coolbho3k, 2022-01-12:
Someone on Reddit said Mystery Gift in Pokemon G/S/C works. Want to try with your setup?

coolbho3k, 2022-01-12:
Here's the LED/transceiver area from the PCB shot from the Analogue OS page. 

The one on the left is clearly an LED, and it seems it's only one LED - with A representing the anode side and K representing the cathode side, but I'm not aware of any small SMD LEDs that can emit both visible light and infrared light, and not only that, at the same time. Maybe I'm wrong though.

The one on the right is a mystery, it's marked with the square wave symbol and I'm not sure what the "E" and "C" near it mean.

[image]

coolbho3k, 2022-01-12:
For it's worth I cannot get the learning remote homebrew to work and and also cannot see anything when pressing a button with my phone camera where it does see my IR remotes.

coolbho3k, 2022-02-02:
Any improvement in 1.0B? I kinda doubt it as it wasn't mentioned in the patch notes, but it would be interesting to try

JoseJX, 2022-06-09:
No improvement on transmitting on 1.0B. I tried with a few things and checked with my Phone camera. Receiving seems to be fine, transmitting seems to be broken still.

RifeXD, 2023-09-01:
Hi I just got my two Analogue Pockets to test with and figured I'd chime in. I'm on the latest version as of posting (on both Pockets) and I can't get Bomberman Max Charabom battles to work. Similar issue, when trying to connect Pocket-GBC or Pocket-Pocket I can sometimes press A and get one or both consoles to show "Processing..." but it's stuck at that screen forever until I cancel.
Will also mention that the placement of the IR is kinda bad. It's impossible to use it while docked so you can't record clean footage of anything being done that uses the IR. Luckily for Charabom battles at least they only need to connect once and then the battle plays out, but just wanted to mention it here.
[image]

RifeXD, 2023-10-09:
ANOTHER UPDATE: Here is the footage with two Pockets: https://www.youtube.com/watch?v=hLL-l-15jRE

RifeXD, 2023-10-09:
And here I'm recording it with the dock. Very finicky, but it works https://youtu.be/hLL-l-15jRE