frederic-mahe/Analogue-Pocket #42: Night of the Living Treehouse of Horror (GBC): Missing graphics and flashing HUD

manksalot, 2022-01-12:
**Pocket firmware:** 1.0A
**Cart condition:** Original and pristine; does not use a save battery, so none to replace
**Problem:** Issues begin when level 1 starts. Graphics are missing and the HUD at the bottom appears corrupted and flashes constantly. 

Using a password to go to level 2 demonstrates the same problems. Have not tested further levels.

The cartridge plays without problems on an AGS-101.

JoseJX, 2022-01-12:
I can confirm that this occurs with my flash cart (Ez-Flash Jr.) and this game as well.

BestPig, 2022-01-12:
I'm also reproducing the bug with a cartridge (Working fine on an AGB-001).

francoiswnel, 2022-08-02:
@frederic-mahe This is reportedly fixed in the 1.1 beta according to the release notes. https://www.analogue.co/support/pocket/firmware/1.1-beta

manksalot, 2022-08-03:
Replayed the first two levels and verified that firmware 1.1 fixes this issue

closed 2022-08-03 by manksalot