frederic-mahe/Analogue-Pocket #44: Gamegear Carts not Booting

Arithmeticae, 2022-01-14:
Some gamegear cartridges fail to load on the Pocket. 

Prior to trying to run the carts on the pocket, the carts were cleaned with deoxit and 90% IPA. Both carts that I experienced issues with ran reliably on a real gamegear. 

Although I suspect this is just the start of the list, the games I experienced issues with question are:
Aerial Assault
Micro Machines

_It is also important to note that the roms for these games work when used through a Gamegear Everdrive X7._

Arithmeticae, 2022-01-14:
Added test point, both carts work fine when used with the MegaSG.

sfcgeorge, 2022-01-14:
Same issue. But after multiple attempts of quitting game and wiggling the cartridge eventually I get it to start, but not sure if that's the repeated tries or the wiggling. Affects Sonic 1 and 2 for me.

Arithmeticae, 2022-01-14:
I spent some time testing my entire collection of GG games and only the ones I've listed above am I unable to get working. 

In my testing, I separated games that had difficulty booting from games that booted first try. Those games with difficulty booting I cleaned with deoxit and IPA after which 100% of them booted first try. This included Sonic 1 and 2. 

I think it's important to separate games that work eventually, even with some wiggling, and games that never boot. Games that require wiggling indicate corroded contacts on the cartridge and I doubt are related to this.

Kypck, 2023-02-15:
Just to add my own tests here:
Aerial Assault looks like new, but wont boot here too. 
Works fine tho with MegaSG and the GG adapter there.

So maybe the Game was generally less quality run or the Pocket has some Issue with some games. Added a picture of my copy. Maybe there is a similarity between copies that do not work. I will get a second one in around a week.
[image]

shawn99452, 2023-02-15:
Not Game Gear, but I actually had one game, Pocahontas (GB), that worked fine in real systems and a Super Game Boy, but required me to re-tin the (slightly pitted) contacts to get it to work in the Pocket, after which point it worked fine.  It's definitely picky.  This has also caused issues with sleep mode for me a few times, where I put it to sleep, it gets minutely jostled while sleeping, and when I resume sleep the cartridge is not perfectly seated and the resume fails.  I've taken to always taking a memory before sleeping, just so I don't lose progress if it fails to resume and I have to reseat the cartridge, and I try to make sure the cartridge is fully seated before resuming it from sleep.

AceHeartTron, 2023-06-03:
I can also confirm from my collection that the original cartridge of Aerial Assault does not boot in the Analogue Pocket.

It is also interesting to note that I have a romset of GG games that includes two variants of Aerial Assault, one identified as 
"World" and a second identified as "World Rev 1".  The Rev 1 rom will work in the openFPGA GG core, but the other will fail to boot in the identical fashion as the original cartridge I have.  There may be a correlation here.

Thanks!

Squuiid, 2023-12-11:
Same issue here. My copy of Aerial Assault on Game Gear looks _brand new_ and won't run in the Pocket.
Works fine in a regular Game Gear.

DJBabyBuster, 2023-12-11:
Very stranger error here to report. My thoroughly cleaned Double Dragon game gear cart loads up and is playable with the Pocket Adaptor, but only character sprites load, no background tiles. The rom loads perfectly on Open FPGA. I have about 15 other gg carts and the rest of them eventually load up. Screen shot of cart glitch and functioning open fpga for reference 

[image]
[image]

shawn99452, 2023-12-11:
Is your ROM from your cartridge, or just a random one from the internet?  Does it work on a real Game Gear?  I suspect if you dumped your cartridge you might have the same issue with its dumped ROM, and the issue is a bad ROM or bad solder joints on the PCB.  I can't find a picture of that game's PCB online but if it only has one chip like most Game Gear games, and all the address and data lines are just wired up directly to it, there's not much outside of the ROM and the connections to go wrong.  Based on how it looks it's probably one of the higher address lines; assuming the tiles are organized in memory with program at the start, and sprites and background tiles separate farther in memory, a high address line being shorted or open could explain the game booting and working, and certain sprites working, but not others.

DJBabyBuster, 2023-12-28:
Original Double Dragon cartridge, single rom chip as most, none of the solder joints looked bad when opening up to clean the contact pins.  Unfortunately don't have any og game gear hardware to test otherwise

stevemassart, 2024-02-24:
My copy of Aerial Assault also not working. Works fine on my Game Gear but not in my Analogue Pocket.

jonknapp, 2024-05-15:
FWIW my copy of Aerial Assault does not run either.

mattculler, 2024-09-08:
My copy of Aerial Assault also does not work on the Analogue Pocket, nor does the ROM I have load up in OpenFPGA.

The Pocket detects the cart correctly and displays metadata about it, but when I hit Play the screen goes blank. 

Running Analogue OS v1.1.

RNekic, 2024-12-30:
I picked up a very clean Aerial Assault cart today and this is still a problem. Analogue OS v2.4 displays the metadata for the game but the game doesn't start. Can also confirm my Aerial Assault rom fails to load via the openFPGA core. I'll have to go find the working rom revision mentioned above.