frederic-mahe/Analogue-Pocket #48: Unable to use IR port in Chee-Chai Alien

Xukkorz, 2022-01-19:
Game Chee-Chai Alien requires you to hold up the IR port to incandescent bulbs to activate certain parts of the game. IR port on Pocket is either not sensitive enough or does not respond to the same wavelengths. Tested on same bulb as on original hardware and where it worked on gbc, it did not work on pocket. On going to light sensing section of game there was a blink where it looked like it saw some light, but then it cut out to nothing.

Game is GBC only and does load, just unable to actually use IR port with game.
https://en.wikipedia.org/wiki/Chee-Chai_Alien

coolbho3k, 2022-01-19:
I haven't been able to get the IR port working as well. See #41 as well :(

GreatHierophant, 2022-01-21:
How well does Chee-Chai Alien work without that "spectrum communicator" attachment?  The attachment is shaped to clip onto a Game Boy Color.

Xukkorz, 2022-01-21:
The spectrum communicator is just a big orange piece of plastic that doesn't even cover the IR port. Not sure it does anything other than possibly reflect a bit more light down at the sensor. Looks cool though.

JoseJX, 2022-06-09:
It seemed to be working for me when using a .pocket conversion and pointing a remote control at it. The IR response reflected my button presses, although they did seem weaker.

Xukkorz, 2022-06-09:
It kinda works but then cuts out I can try and film a video but it seems like it just cuts out.

JoseJX, 2022-06-09:
Maybe the IR sensor isn't as sensitive to the light bulb's IR frequency? I wasn't using a light bulb, just a regular Visio remote control.

Xukkorz, 2022-06-09:
Possibly, gbc was a very different response but it was like it got ghost input on Pocket then didn't do anything more.  I'll try to film it and post the difference in the morning.

coolbho3k, 2022-07-30:
Any better on 1.1 beta?

Xukkorz, 2022-07-31:
No change in behavior.