frederic-mahe/Analogue-Pocket #51: Mortal Kombat Deadly Alliance and MK Tournament Edition do not load correctly and are glitchy and unstable

JBGV, 2022-01-25:
Authentic carts, both work on original hardware. Tried them in my Pocket, and they both stay on the Midway logo for about a minute, and the logo does not animate. Eventually the games load, but various elements are the wrong color, text is nonsense characters, and the games are unstable and crash prone.

 

https://imgur.com/a/23Dni9k

Mikemetroid, 2022-01-25:
Hi, I'm the OP of the reddit thread and wanted to come by to verify the validity of this post. Please let me know if you need any sort of info from me. Thanks

Mikemetroid, 2022-02-03:
Thank you! This has been resolved with firmware 1.0B

frederic-mahe, 2022-02-03:
@JBGV please confirm that the problem has been solved, and close this issue.

JBGV, 2022-02-04:
I will check the carts tomorrow! Thanks.

ZReport, 2022-09-08:
This issue seems to have been resolved by the v1.1 beta 3 patch.

Mikemetroid, 2022-09-08:
We are definitely good. You guys fixed this a while back.

Thanks for the reply,
Mike