frederic-mahe/Analogue-Pocket #52: Green lines in Cubix: Race’n’Robots

zxtheproto, 2022-01-26:
**Issue**

When playing “Cubix: Race’n’Robots” for Game Boy Color, green lines appear regularly during the game’s dissolve effects. (This issue also occurs when playing this game on the GBA SP, albeit to a lesser degree.)

**Examples**

Game running on a stock Game Boy Color:

https://youtu.be/Ehe3qXq-V4E

Game running on the Analogue Pocket:

https://youtu.be/_u6ymmKCDVA

zxtheproto, 2022-01-26:
Something else I’ve noticed about this title is that the pan of Cubix on real hardware aliases slightly as it moves, whereas all other hardware I’ve tried seems to present it as a smooth up-down movement. Not necessarily a negative, but different nonetheless.

GreatHierophant, 2022-01-27:
I see the same kind of lines on my GBA and GBA SP AGS-101 as I do on Pocket, so it is possible that some unique property of the GBC's screen is keeping them from being visible.

zxtheproto, 2022-01-27:
Have you been able to replicate them in the menus or gameplay? There is a similar looking effect that occasionally happens on the black screen before it says “Licensed by Nintendo” on the GBC screen, but I have never been able to replicate it in any other area.