frederic-mahe/Analogue-Pocket #55: Gb modes

mddifilippo89, 2022-01-31:
What's your thoughts on having all the available modes for GB?

Force GB
Force SGB
Force GBC
Force GBA - Sounds weird, but I am sure someone wants to play gb carts as cropped widescreen as played on a real gba when using the triggers.

16bitstreamer, 2022-01-31:
That's not what Force GBA does. There are a few GBC games that take advantage of the GBA's power to do extra things, Shantae for example. This allows those games to use those advantages where available. 

The options all have significate use, and I appreciate them greatly.

bender-the-greatest, 2022-02-01:
Should we be using this tracker to discuss feature wishlists? This isn't an official issue tracker maintained by Analogue.

16bitstreamer, 2022-02-01:
I feel that the primary purpose for this tracker is to track issues and bugs.

closed 2022-02-01 by mddifilippo89