frederic-mahe/Analogue-Pocket #56: Legend of Zelda LTTP Four Swords Graphic Glitch

16bitstreamer, 2022-02-01:
Noticed this after updating my dock to Firmware 1.0B, there appears to be purple dots flickering in areas. Not present when playing handheld, only visible when docked. 

Happens in 1080p & 720p. I have video and screenshots but they aren't uploading from my phone.

16bitstreamer, 2022-02-01:
[image]

coolbho3k, 2022-02-01:
Also happening in other games:  https://www.youtube.com/watch?v=E1lRG34ITEA

coolbho3k, 2022-02-01:
Mind changing the title to include the fact that it seems to occur in docked mode on 1.0B on more than one game? Maybe `GBA docked visual artifacts on 1.0B`

horntm, 2022-02-03:
It seems a few Redditors are also having this issue. I went ahead and sent Analogue Support an email along with links to some video I captured displaying the pixel flickering issue. Hopefully the next update irons this out.

PassPose, 2022-02-04:
I've also has this issue. While it's most prevalent on the GBA I've also noticed it to a lesser extent on some GBC games, like Warioland 3.
[video on Twitter](https://twitter.com/AndyPalatka/status/1489364732150640640?s=20&t=-x8oxnWWgpumoUOjnV_kaA)

blinky36, 2022-03-11:
It's very prevalent in Mario & Luigi: Superstar Saga

tromero, 2022-05-07:
Occurs quite visibly on the title screen of GBA Super Mario Advance 4: Super Mario Bros 3. Also occurs in GBC Pokémon Trading Card Game

Eatitup86, 2022-06-25:
This should be their main priority imo... and really should have been hot fixed months ago. Dock is unusable while this issue is there as it happens on nearly every single GBC and GBA game.

coolbho3k, 2022-07-30:
Seem fixed with 1.1 beta