frederic-mahe/Analogue-Pocket #59: Analogue Pocket Dock: GBA Pixel Flickering in Firmware 1.0b

horntm, 2022-02-03:
When in docked mode, GBA titles started exhibiting flickering pixels since the 1.0b update. I tried it on Tales of Phantasia, Mario Kart: Super Circuit, and Metroid Zero (all original GBA carts) in all display modes and resolutions. When undocked, the GBA games display normally on the Pocket. 

Gameboy titles do not display this pixel flickering at all either docked or undocked and display normally.

Again, the GBA pixel flickering issue in docked mode only showed up after updating to firmware ver. 1.0b. I also captured video of this flickering but had to resort to screenshots here. I'll be sharing the video in my support ticket to Analogue.

[image]
[image]
[image]

16bitstreamer, 2022-02-03:
Duplicate to issue #56

https://github.com/frederic-mahe/Analogue-Pocket/issues/56

horntm, 2022-02-03:
My apologies. This appears to have already been reported: #56.

closed 2022-02-03 by horntm