frederic-mahe/Analogue-Pocket #63: [1.0B] EZ Flash Omega Definitive Edition: Bad Header and NOR flash issues

coolbho3k, 2022-02-08:
1. Some users get "Bad Header" on this flashcart on 1.0B. Reverting to 1.0A fixes the problems for these users. ([Example 1](https://twitter.com/JoeSpaghet/status/1488622060200992779) | [Example 2](https://github.com/frederic-mahe/Analogue-Pocket/issues/1#issuecomment-1032167982))
2. Booting from "Mode B" (directly booting to NOR flash) and boot from NOR flash from the menu are still not working

Interestingly, 1.0B actually solved the problems with this flashcart for other people.

Ninbura, 2022-03-31:
Just want to point out that sleep mode seems to work when booting from "Mode B" / standalone mode. Loading in standalone mode results in a white screen as mentioned in previous issues, but you can sleep and wake on this screen without the "game" automatically closing. Where-as if you boot into the regular mode where you'd typically select a game the Pocket warns you that "this game doesn't support sleep". 

The EZ Flash Omega Definitive Edition and the pocket are a match made in heaven if this is fixed. A flash cart that is compatible with one of the Pocket's most convenient features - sleep.

coolbho3k, 2022-07-30:
Both issues seem fixed with 1.1 beta.

closed 2022-07-30 by coolbho3k

nfoste82, 2022-07-30:
On 1.1 beta I get a "Bad Header" when using Mode B on the EZ Flash Omega DE. Mode A works fine. Anyone know of a fix for this?

Ninbura, 2022-08-06:
> On 1.1 beta I get a "Bad Header" when using Mode B on the EZ Flash Omega DE. Mode A works fine. Anyone know of a fix for this?
Do you still get this issue when the pocket is plugged in and charging? See the issue I just created above this reply for more information.

rothomson, 2022-08-10:
Thanks for posting! Didn't think to try it after 1.1. I was disappointed when Mode B didn't work. So I just loaded Pokemon Gaia to NOR and I'm testing now. Seems to be working beautifully (so far) had my save on ice for ages, great to pick it up again.