frederic-mahe/Analogue-Pocket #69: Graphical glitch in Densha de Go! route selection menu

francoiswnel, 2022-02-13:
I've found a minor but reproducible graphical glitch in the Densha de Go! route selection menu.

#### Expected Behaviour

The route selection menu should show four square train icons.

#### Current Behaviour

The route selection menu intermittently shows glitched icons. When the glitched icons show, they are consistently glitched in the same way, i.e. there is no variation in the glitching. This intermittent behaviour happens about 1 in 5 to 1 in 10 times that the route selection menu is opened. See the photo below.

#### Steps To Reproduce

Repeatedly open and close the route selection menu until the glitched menu appears. This may take a few tries, but seems to happen around 1 in 5 to 1 in 10 times.

#### Environment

Analogue Pocket, Analogue OS 1.0b, both physical cart and .pocket conversion.

#### Possible Reason

The menu icons are animated as the menu opens, sliding from the inside out. I think the animation is not completing successfully, leaving the icons in a partially glitched state. Might be a timing issue.

#### Other Notes

I've attempted to reproduce the glitch on an AGS-101 and CGB-001 without success. Interestingly, the CGB-001 appears to play another audio track in the route selection menu. This track is missing on both the AGS-101 and Analogue Pocket. Since this happens on original hardware too, I don't consider it to be a bug. Might log it as an enhancement in the future, to replicate CGB behaviour instead of AGS/AGB behaviour.

[image]

francoiswnel, 2022-07-30:
Have retested this on Analogue OS 1.1, and unfortunately it’s even worse. The glitch appears more often now, even more frequently than the correct graphics, and it’s more corrupted than before.

[image]

francoiswnel, 2022-09-07:
Confirmed that the bug is still present in 1.1 beta 3.

ZReport, 2022-09-08:
> Confirmed that the bug is still present in 1.1 beta 3.

You should change your settings so that the System setting for GBC has "Run as GBA" turned off, then change the System setting for GB to be "Auto-Detect".  Do both of these and check back.

Settings > Pocket > Systems > GB & GBC.

francoiswnel, 2022-09-08:
> > Confirmed that the bug is still present in 1.1 beta 3.
> 
> You should change your settings so that the System setting for GBC has "Run as GBA" turned off, then change the System setting for GB to be "Auto-Detect". Do both of these and check back.
> 
> Settings > Pocket > Systems > GB & GBC.

Run as GBA is off and Auto-Detect is on.

francoiswnel, 2023-12-07:
Can confirm that this bug is still present in 1.2.

francoiswnel, 2023-12-24:
Can also confirm that this bug is still present in 2.0.

francoiswnel, 2024-05-01:
Just posting into the void that this issue is still present in firmwares 2.1 and 2.2. I guess it won't get fixed at this point.

francoiswnel, 2024-09-24:
It's still present in 2.3. The corruption is consistent and I can't get a non-glitched menu to load. It's slightly different in appearance to the last one posted for 1.1.

hackgrid, 2024-09-24:
Have you tried running it on openfpga-GBC?

francoiswnel, 2024-09-24:
> Have you tried running it on openfpga-GBC?

Nope. I don't have any openFPGA cores currently set up on my AP. I use it with physical carts. But that would be a good test if someone is able to check.

hackgrid, 2024-09-24:
You tested with the .pocket conversion, so why not with the original ROM?