frederic-mahe/Analogue-Pocket #72: Robopon - Sun Version locks up

JoseJX, 2022-02-21:
Robopon - Sun Version (U) crashes after showing the Hudson Logo with both a patched .pocket file and when using a flash cart.

JoseJX, 2022-02-21:
This is probably due to the HuC-3 mapper used for this cart. I'd expect that the other carts that use this mapper would also exhibit issues.

manksalot, 2022-02-21:
I have an original US cartridge and it loads past the Hudson logo, I can create a new game and play. So far functioning normally.

JoseJX, 2022-02-21:
Okay, interesting! So maybe it's a mapper issue with flash carts too? It was tested with an Everdrive by a user on the classic gaming Discord. So, perhaps we'll need to work around it another way then.

Thanks for testing with the original cart!

BestPig, 2022-02-22:
I checked with my EverDrive on a GBA and the game is recognized as a MBC5 and it fails to run.
Also fails on real GBA with the game flashed in a bootleg.

So it's not an Analogue issue, just flashcard are not HuC-X compatible nether the GB Studio mode.

JoseJX, 2022-02-25:
Okay, so this isn't a Pocket bug, thanks!

closed 2022-02-25 by JoseJX