frederic-mahe/Analogue-Pocket #73: Pokemon Emerald save file corrupted after using sleep mode

mattshynn, 2022-02-22:
I played a fresh save on a legit copy of Pokemon Emerald for around 7 hours. Used a mix of sleep mode and turning the pocket off completely, but most recently left it on sleep mode for a few weeks. Went back to play the game and got approximately 5 minutes in (no saves in this time) and the game locked up. Rebooted the console and got the 'the saved game has been deleted due to damage or corruption' message. I've dumped the .sav file using a GB operator and attached here in case it helps.

[Pokemon - Emerald Version (USA, Europe).zip](https://github.com/frederic-mahe/Analogue-Pocket/files/8118259/Pokemon.-.Emerald.Version.USA.Europe.zip)
.

4wilko, 2022-03-21:
Did this happen with the 1.0B firmware?

mattshynn, 2022-03-25:
Sorry, i meant to say this was with 1.0A - is this supposedly fixed in 1.0B?

4wilko, 2022-03-26:
> Sorry, i meant to say this was with 1.0A - is this supposedly fixed in 1.0B?

One of the listed changes was "Wario Land 3 Sleep/Wake issue - Fixed" and I've been unable to reproduce the save loss issue due to sleep mode I was having with a different game since I updated. I'd recommend giving it a try.