frederic-mahe/Analogue-Pocket #85: Color banding on dock HDMI output when using capture equipment

Ninbura, 2022-05-12:
I've noticed that there's very distinct color banding on the Analogue Pocket Dock's HDMI output when interfaced with capture equipment. 

It's most noticeable when the screen is displaying a single color, but is noticeably displayed across many environments. Here's the most clear example that I've found, the home screen of EZ-Flash's [Omega Definitive Edition Flash Cart](https://www.ezflash.cn/product/ez-flash-omega-definitive-edition/):
[image]

Here's that shot but zoomed in:
[image]

The above was captured using Magewell's [Pro Capture Dual HDMI 4K Plus LT](https://www.magewell.com/products/pro-capture-dual-hdmi-4k-plus-lt) in OBS with these settings:
[image]

XRGB is the highest format quality, and from my understanding should be essentially lossless... yet the image does not mimic what I see on screen, which is zero color banding of any kind. I've also tested the signal with Elgato's HD60 Pro, and 4K60 Pro and get the same results, color banding that is. I've continued to test across multiple PCs, with multiple cables, and with the help of a friend; multiple Pockets and Pocket Docks. The problem is consistent across the board.

Interestingly I've found that if I upscale the signal using the Magewell capture card, which supports upscaling and downscaling internally, the banding goes away. The problem with this is now I have to pull a 4K signal which is more intensive, and XRGB at 4K is not possible. Still the image is much cleaner, even with NV12 as the format. Shortly after with more testing I've found that I get the best image when I use HDFury's [Integral 2](https://hdfury.com/product/integral-2/) to upscale the image to 4K, pipe that into the Magewell capture card, and then downscale it back to 1080p. This way the Magewell is doing scaling, which clearly has some signal cleanup process, and since the final signal is 1080p I can use the XRGB format. Here is the result:
[image]

I know the Magewell card is doing some sort of cleanup while scaling because if I downscale using the Elgato card, which only supports down scaling, there is still banding. So same thing - Pocket Dock > Integral 2 upscale > Elgato downscale - still results in banding. I've put together a Figma document that displays the signal across various capture cards with various settings, I also show that the problem persists in retail games / authentic game carts, if that was in doubt:
https://www.figma.com/file/1HAEdUCnklTJNzw9pjryVl/Analog-Pocket-Color-Banding?node-id=0%3A1

Disabling Figma's pixel grid makes the banding much easier to see when zooming in, this setting can be found int he top right corner of UI:
[image]

While I've found a pseudo-solution with the combination of my HDFury Integral 2 and Magewell Capture card, is a pretty unreasonable solution... $2000 of professional video equipment. I typically dedicate my Magewell capture card to my main PC display, I'd much rather offload this to my Elgato capture card which typically does great with standard 16:9 RGB images. I assume this issue has something to do with the conversion of the color space in some way, though I'd think with XRGB that wouldn't be an issue. I've gone on to test the exact same colors by loading images onto my Nintendo Switch and PS5, but there is no banding. So the issue seems to be exclusive to the Pocket / its dock. 

Fixable? Or is this just the nature of the original Game Boy's signal or something? Any insight appreciated.

Ninbura, 2022-05-12:
Pocket & dock is running the latest firmware (1.0B), so was my friends.

mddifilippo89, 2022-05-27:
thank you for your post. you got to send an email to analogue about all this and press them. The dock is a huge reason I got the pocket and it needs just as much attention. We all want the best presentation possible regardless if it's the pocket or dock.

GreatHierophant, 2022-07-24:
I cannot perceive any banding with Kingdom Hearts and Dock.  This capture using my Datapath VisionRGB E1s capture card at 480p :

[image]

Ninbura, 2022-07-27:
> I cannot perceive any banding with Kingdom Hearts and Dock. This capture using my Datapath VisionRGB E1s capture card at 480p :

This is consistent with my findings... you are scaling the image at some point. You capture card is likely performing some level of color correction as is mine.

Try capturing at the native 1080p resolution of the dock.

GreatHierophant, 2022-07-27:
My capture card was doing a 2x scale, but the Dock was outputting in 480p mode.  Here is a screenshot of the menu in native 1080p mode, no scaling:
[image]

Ninbura, 2022-07-27:
> My capture card was doing a 2x scale, but the Dock was outputting in 480p mode. Here is a screenshot of the menu in native 1080p mode, no scaling: [image]

I don't see any banding in this image either, but there is still something weird here. The game image isn't extending to the border of the screenshot, the game image itself is only 1440x960.

What program are you using to capture the frame, and with what settings?

Ninbura, 2022-07-27:
Could you try capturing a screenshot without the fixed scale setting?

GreatHierophant, 2022-07-27:
[image]

There is interpolation in the stretch on both axes, but the sharpness is set to 3.

Ninbura, 2022-07-27:
Huh... I'm not sure why there isn't banding.

I'm interested to see if I get no banding with fixed scaling, I'll have to test later. I'll also need to try capturing screenshots with FFmpeg to make sure it's not just an OBS problem.

There is a noticeable difference in colors and artifacting when comparing our images:
[image]

Ninbura, 2022-08-06:
This issue has been fixed on firmware v1.1-beta.

closed 2022-08-06 by Ninbura