frederic-mahe/Analogue-Pocket #91: No cartridge inserted - ERROR: BAD HEADER

TheMostSecret, 2022-08-04:
Getting a new error with the 1.1beta firmware. When I select play cartridge when no cartridge is inserted I'm getting the error:
ERROR : BAD HEADER
_
CLEAN CARTRIDGE PINS

Thought at first maybe my cart slot had went bad and was giving errors but I downgraded back to 1.0b and the error cleared itself up. Biggest issue with this error is that it prevents the use of Multiboot/Link Mode as the system always thinks a cartridge is in the slot.

ZReport, 2022-08-04:
What game is it that you're trying to play?

TheMostSecret, 2022-08-04:
> What game is it that you're trying to play?

Not trying to play a game on the Pocket itself, I was trying to use the Tingle Tuner in Wind Waker on Game Cube with the link cable. However the Pocket THINKS a game is inserted and won't activate Link Mode, gives the error above , and the Game Cube says no GBA is plugged in.