frederic-mahe/Analogue-Pocket #93: [GBC] (OpenFPGA) Jurassic Boy II (Unlicensed) won't boot

italiangrandma, 2022-08-05:
Booting the game results in "Error" message

TheMostSecret, 2022-08-05:
Unless this is an issue with the physical cartridge you might want to add this to the openFPGA-GB-GBC git instead.
I've already got an issue open there about unlicensed mappers if you want to add this game to the list.
https://github.com/spiritualized1997/openFPGA-GB-GBC/issues/10

italiangrandma, 2022-08-06:
@TheMostSecret Thanks for the heads up! I actually wasn't aware we could add issues on that page. That said, I tested this on the ED-GB flashcart as well and experienced the same behavior. Not the original cart, so maybe I should still move it?

TheMostSecret, 2022-08-06:
@italiangrandma
No problem, I'm really hoping more mappers get supported on that core so the more info there the better.
As for the ED-GB, it still has to emulate the mapper as well so it's not a great baseline. It is entirely possible that the actual cart wouldn't boot either (and given that it's an unlicensed game the odds are good that it wouldn't) but with out being able to test it we can't say for sure.
I'd say just keep this open just in case someone does have the actual cart to test out.

tjanas, 2022-08-09:
If this occurs with both the anonymous unofficial [openFPGA-GBA](https://github.com/spiritualized1997/openFPGA-GBA) core as well as an EverDrive cart, then this issue should exist along with an issue created in the openFPGA-GBA repo. @italiangrandma can you create an issue ticket over there?

italiangrandma, 2022-08-15:
@tjanas Sorry for the late reply! I think you meant to say the openFPGA-GB-GBC repo since this is a GBC Game. I did add this to this issue: [Unlicensed and obscure mappers not supported](https://github.com/spiritualized1997/openFPGA-GB-GBC/issues/10)