frederic-mahe/Analogue-Pocket #97: Beta firmware 1.1 - Creating or loading save state for The Lost Vikings causes audio to disappear

beckbria, 2022-08-17:
When playing The Lost Vikings (Gameboy Advance), creating or loading a save state/memory causes all audio to drop until the cartridge is restarted.  This is the only game that I have found this issue with - others work fine.  Also appears to repro on Everdrive GBA.

shawn99452, 2022-11-28:
This happens with Blackthorne (GBA) also, with an original cartridge; probably they use a similar audio engine, since both are made by the same developers.  In Blackthorne, the audio eventually comes back if you finish a level, but dies again if you create / load a save state.  Sleeping the system also triggers the issue, I think because it "sleeps" by creating an interim save state and then loading it on wake.

seanfcastillo, 2022-12-18:
I’m seeing the same issue in Legacy of Goku (original cartridge) and legacy of goku 2 (openfpga). 

Sound gets broken on sleep/wake and also on save states. Only way to fix it is to restart game. 

Weird that it affects these games too.