frederic-mahe/Analogue-Pocket #100: Action man on gbc not working since beta 1.1 version 1

Beldandy561, 2022-09-09:
I am writing to submit an observation.

Ever since Beta firmware 1.1 version 1 has been released compatibility  with Action Man on the gameboy color has been broken. (Still broken in 1.1 beta3)

booting any version of the rom cycles  through it's original company logos and then crashes at a blank white screen when it tries to boot the title screen.

 I have tested this with each version Of the 1.1 Beta firmware in the following environments.

.pocket version of all 4 variants of the rom with the same issue and results.

 Known working Chinese knock off of a Ktizz Gb/Gbc flashcart that did work with either of the 4 versions of the rom just fine before fw beta 1.1 and on., now crashes with the same results as listed above.

 Using all 4 rom variants with the spiritualized core and original GBC bios with the same results.

 The only thing I do not have access to is the original cartridge.

Otherwise all of these attempted tests have resulted in identical  crash results to a white screen after the company logos.

MetalVinnie, 2023-02-06:
I can confirm this issue AND that this issue is also happening with the physical cart, which I do own.

Beldandy561, 2023-02-07:
Thank you for the heads up we will just have to wait to see if analog ever
resolves this.