frederic-mahe/Analogue-Pocket #103: Worms Armageddon (GBC) is very glitchy. Water not rendered, tiles loading incorrectly.

advert665, 2022-11-22:
Worms Armageddon on gameboy colour is very glitchy on the AP. The water does not load, and scrolling around the map using the select key causes graphical glitches with tiles loading in incorrectly. Image 1 below shows the worm drowning where water should be rendered. Image 2 shows the tile glitches that occur when you scroll using select.

[image]
[image]

andreevgy, 2024-04-11:
Can confirm it as well, happens to me with original cart. Does not happen on FPGBC though.

arhneu, 2024-05-15:
Disappointingly, this is still an issue, two years on. Especially with the game recently re-released officially, it's a shame...

CommanderGrog, 2024-11-03:
Yes, that's true. Especially since the game is running flawlessly with the available OpenFPGA GBC-cores. I'd really love to be able to play my real cartridge, though, since this is, what I bought the pocket for, and since lots of memories are connected to it.