frederic-mahe/Analogue-Pocket #104: Some GBA titles fail to save with open FPGA

catnipped, 2022-11-25:
Of the ones i’ve tried, both Final Fantasy Tactics Advance and Tactics Ogre - Knights of Lodis ”fail to save” when trying to save on their respective screens. This is even if i manually added a save file to the save folder (it loads but i can’t save)

This is on the latest firmware 1.1b6

sopesitos, 2022-11-26:
Apotris doesn't seem to save as well

https://akouzoukos.itch.io/apotris

akouzoukos, 2023-01-13:
For Apotris, I had to add the [SRAM_Vnnn string](http://problemkaputt.de/gbatek-gba-cart-backup-ids.htm) in the code for saving to work.

SRAM should probably be the default save type if no save type was detected, unless there's something preventing this. Most GBA homebrew supports SRAM and I'd imagine those are the games where a save type wouldn't be detected most of the time.

GreatHierophant, 2023-03-30:
I can save with both games' USA versions with openFPGA.