frederic-mahe/Analogue-Pocket #113: Tumblepop-USA: scrambled screen after beating any level

MetalVinnie, 2023-02-06:
I have a DMG-T6-USA-1 Tumblepop cart for Game Boy currently using firmware 1.1-beta-7 on the AP.

Playing the game on the AP is okay until you beat one of the levels. The map will be scrambled after that.  If you go to a bonus round after the first level, that will also be scrambled. You can exit to another map or play another level, and the scrambling will go away until you beat the level and it will scramble again.

I know there are 2 revisions of the USA version, so for now I'm just posting about the revision I have (USA-1)

Hard to determine which revision it is on the USA roms circulating out there, but the same issue applies to the roms I've tried out so far. I have tested the rom using an Everdrive GB X7, I have tried running the rom on the third party FPGA core, and I patched the rom to work with GB Studio.  All iterations have the same problem.

I did test the Japanese region rom and no problems with that. So, thankfully there is a way to play it properly.

But it would be nice if this rare and expensive USA cart can work with the Analogue Pocket. Hopefully I'm sending the message to the right people. Thanks in advance.

16bitstreamer, 2023-02-06:
[image]

Have tested and can confirm issue. 

Works without error on the MiSTer. (See below, same conditions.)

[image]

MetalVinnie, 2023-02-23:
There are reports, now, that there is a problem with both the USA revisions (Data East or Sunsoft)