frederic-mahe/Analogue-Pocket #119: Support of bootleg pokemon cartridges with SRAM

tuman18, 2023-05-20:
Since the firmware versions above 1.1-beta-2 I cannot get the Pokemon game working right after transition to save selection screen. [Here is what happens ](https://www.youtube.com/watch?v=QFxa8jfTFac)

The problem with these newer bootlegs is that they use SRAM memory and not the FLASH. Saving mechanism is also different, as the save data is written directly to the ROM file. Here is [one example of this type of cartridges ](https://i.imgur.com/L799znc.jpeg), and mine is a bit different using the Samsung memory chip.

tuman18, 2023-08-28:
@Xoliul Are they new too, using SRAM?

Xoliul, 2023-08-28:
If I open a GBA Firered bootleg, there's no battery, it seems to have a Spansion flash chip.
Same for a GB version, also no battery and a different Spansion chip.

Xoliul, 2023-08-28:
The Firered board had some ST ic chip on it. Yes, I do use the latest version. 
This might be a case where bootleg carts aren't officially supported and your mileage might vary... Do they work in a real gameboy?

tuman18, 2023-08-28:
Yep, the real Gameboy play it without issues