frederic-mahe/Analogue-Pocket #125: GBA - Pokemon Emerald save file gets corrupted after using Memories

karlphillip, 2023-10-07:
I using the latest Analogue Pocket firmware which at this time is **1.1** [2023-05-04]. 

The problem happens after creating a _Save State_ (Memories) and loading it, then trying to save the game using its built-in _Save_ feature will result in a `.SAV` file stored in `Saves\gba\common\` that is filled with `0xFF`.

I can only speculate that the issue appears to be caused by either _Memories_ not being able to create a proper _Save State_ or even a bug caused by the procedure that incorrectly loads a _Saved State_.

I lost many hours of progress on Pokemon Emerald and won't be investing in additional Analog products until this issue is fixed. The Glow in the Dark and Transparent editions look great though.

SaulGoodman1337, 2023-10-08:
same behavior here with another game

guillaumeslash, 2023-10-31:
Got a pokemon blue save corrupted the same way on openFPGA (not a cartridge)

davidretler, 2023-11-27:
Same issue with Shining Soul II via OpenFPGA. Any word from Analogue on this?